Concept-Bytes/Tipsy #2: build plans or guide?

pilot8251w, 2025-03-27:
I have looked all over and can't find a guide to assemble the parts. There are a few parts that I'm not sure where they go.. Most are self-explanatory. do you have an assembly guide somewhere?

ipatiris, 2025-03-27:
I have asked the same though thangs. No reply so far.

EricHigdon, 2025-04-12:
@pilot8251w @ipatiris You can get the build manual and the 3d print files here: https://concept-bytes.com/products/tipsy-build-manual
The build manual is pretty high-level, so you'll need to be able to understand how to do a lot of problem-solving yourself.

I'm currently working on building this, and have it mostly working (just waiting on the pumps to arrive) so if you have questions, I may be able to help.

mehmettem, 2025-06-30:
@EricHigdon Did you power your Waveshare screen through the Pi HAT as well? I’m new to electronics, so I thought the DSI cable would provide both power and signal to the screen. However, my screen doesn’t power on with just the DSI cable; I have to use a separate power cable from the Pi HAT. If an additional power cable is needed, is using the Pi HAT the best way to do it, or did you use another method or location for powering your screen? Thanks for your help in advanced.