Concept-Bytes/Tipsy #10: Ghost Trailing on Interface.py

elated-emu, 2025-04-18:
Ghosting Effect on interface.py

[image]

Pyrogenez, 2025-04-21:
Thank you for posting this!! I thought something was wrong with my screen so I returned it only to realize it must be the programming.

kcoda36, 2025-04-22:
Which RPI are you using?

elated-emu, 2025-04-22:
> Which RPI are you using?

Tested on Raspberry Pi 5 16gb, RPi 5 8gb, and on another laptop to make sure my screen wasn't faulty. Happens on every device

Pyrogenez, 2025-04-22:
> Which RPI are you using?
> 

Raspberry pi 4B 

The background and the single/double icons have trouble with trails too. 

I've tried all today and last night to fix the issue but can't figure it out.

elated-emu, 2025-04-22:
It also seems like the background is trailing and "cloning" over itself

elated-emu, 2025-04-22:
https://github.com/user-attachments/assets/1bf95c3c-0c78-482f-9116-ba9d1de744b2

Hope this helps

Pyrogenez, 2025-04-29:
> https://github.com/user-attachments/assets/1bf95c3c-0c78-482f-9116-ba9d1de744b2
> 
> Hope this helps

@kcoda36 @EricHigdon - any tips to overcome this problem?

EricHigdon, 2025-05-17:
This happens because the background image has an alpha layer, so you are seeing previous frames through the current frame.
The easiest solution is to remove the alpha layer from the background image.