Concept-Bytes/Tipsy #12: Problem with pump operation

Gilzano, 2025-04-21:
I'm having a problem where the pumps work when I click clean in the web interface but when I select a cocktail (and not in debug mode of course) no pouring takes place
I'm working with a Raspberry Pi 4

Gilzano, 2025-04-23:
> I am having the same issue on RPi5
> 
> It seems like selecting a drink from the web interface also allows you to pour. However the main interface is bugged

It seems that interface.py does not have a call to controller.py at all.

ipatiris, 2025-04-23:
Hello @Gilzano  @elated-emu @EricHigdon @kcoda36 I have made all the necessary changes in order to make this happen. I have created a pull request for merging to the main. Meanwhile you can check my forked version.

ipatiris, 2025-04-24:
@

> It seems that [@EricHigdon](https://github.com/EricHigdon) solved it and the pumps are even pouring at the same time Just now it doesn't show the image of the pouring but that's not bad. The window doesn't open as full screen but it's a simple solution

I was planning to do next parallel pumps but if @EricHigdon has done it already thats perfect. Another think that can make the pumps faster is powering them with 12V but not through Tipsy hat since is powering the raspberry as well(5V). Ofcourse this makes the need to have two power supplies (12V,5V).

What kind of pumps have you bought? I had a look on Aliexpress and found some with around 5 euros. Not sure if dimensions are exactly the same....

elated-emu, 2025-04-24:
I'm getting a 12v power supply and a 5v regulator for the pi tomorrow.

Using a relay to power the motor drivers connected to the 12v. Will update back with any power issues.

Gilzano, 2025-04-24:
> @
>
>
> I was planning to do next parallel pumps but if [@EricHigdon](https://github.com/EricHigdon) has done it already thats perfect. Another think that can make the pumps faster is powering them with [...]

It seems a bit dangerous to combine the 2 voltages (5v and 12v) when there is a common circuit between them
In any case, these are the pumps I bought:
https://s.click.aliexpress.com/e/_ooabIct

ipatiris, 2025-04-24:
> It seems a bit dangerous to combine the 2 voltages (5v and 12v) when there is a common circuit between them In any case, these are the pumps I bought: https://s.click.aliexpress.com/e/_ooabIct

What do you mean about common circuits and what about combine the 2 voltages? The only think to pay attention is that they should have common ground.
What I meant is 12V goes to pumps though Tipsy hat. Tipsy hat pins that are normally powering the raspberry pi need to be removed from hat (Pins 2 and 4).

Gilzano, 2025-04-24:
> What do you mean about common circuits and what about combine the 2 voltages? The only think to pay attention is that they should have common ground. What I meant is 12V goes to pumps though Tipsy [...]

Yes I understand you
If you cut the pins leading to the Raspberry Pi it's fine. I still don't like the idea and my pumps are 6V

Pyrogenez, 2025-04-26:
> I'm getting a 12v power supply and a 5v regulator for the pi tomorrow.
> 
> Using a relay to power the motor drivers connected to the 12v. Will update back with any power issues.

Any update on this? I was thinking of doing this too.  I've already purchased the 12v version of the recommended pumps.

elated-emu, 2025-04-26:
> > I'm getting a 12v power supply and a 5v regulator for the pi tomorrow.
> > Using a relay to power the motor drivers connected to the 12v. Will update back with any power issues.
> 
> Any update on this? I was thinking of doing this too. I've already purchased the 12v version of the recommended pumps.

I actually just got it working at 2am.

I used this power supply:  https://amzn.eu/d/9myJxFG
These motor drivers: https://amzn.eu/d/fz1CQTo

10 of 12 pumps can be used simultaneously at 12v I think.

If you get a bigger PSU you can push more.

Positive of PSU goes to all positives on all driver boards.
Negative goes to all negatives on all driver boards plus a negative to a ground pin on RPI.

Then wire gpio pins to each board.

I will try expanding to 24 motors later

Gilzano, 2025-05-05:
Although it's not related to the pump problem, does anyone know how to create a button that clears the recipes and images for new use with new ingredients?

EricHigdon, 2025-05-05:
@Gilzano, the streamline app does that now. It replaces all of the cocktails every time you generate new recipes. 
In my fork, I made that optional. So you can add to the existing menu, or start from scratch.

Gilzano, 2025-05-05:
> [@Gilzano](https://github.com/Gilzano), the streamlit app does that now. It replaces all of the cocktails every time you generate new recipes. In my fork, I made that optional. So you can add to the existing menu, or start from scratch.
 
Nice ! Thanks

Hemuc71, 2025-05-18:
> Amazing!
> For now, I just replaced the Interface.py file and added Button.py and the settings image and it seems to be enough to work, although it takes time from the moment it registers pouring to the moment it starts pouring
> but hey at least it works! Thank you very much!
> Is there a way to make the pumps work in parallel to reduce the preparation time?

@Gilzano Hi experiencing the same problem, you mention replacing the interface.py file and the Button.py file. But whth what did you replace them?

EricHigdon, 2025-05-18:
@Hemuc71 he was referring to copying files from my fork of this repository. 
These particular changes have since been merged to the repo. 

Have you pulled the main branch recently? If so, can you be more descriptive of the problem you are facing? I may be able to help.

Hemuc71, 2025-06-09:
@EricHigdon @Gilzano Thanks all, just played around with it started from scratch and got it working with a new fresh branch Eric, thank you very much.