Concept-Bytes/Tipsy #15: Mapping of pumps

ipatiris, 2025-05-10:
During debugging I noticed that mapping of the pumps is incorrect. I think this is because of PCB layout.
1. In the code pump 9 is controlling PCBs pump 11
2. In the code pump 11 is controlling PCBs pump 9
3. In the code pump 12 is controlling PCBs pump 8

Pyrogenez, 2025-05-14:
@ipatiris - I am confused about how to connect the supplied PCB, RPi, and motors.  I inserted the PCB (hat) into the pins of the RPi and then connected by wire, the respected numbers above (A, B), to the terminals on the back of each motor.  Am I missing something? Do these motors require a connection to the power source on the PCB?

Pyrogenez, 2025-05-14:
Oh, wow, thank you!! How are you powering through J14? What type of adapter or plug are you using from the wall to the two points/one port at J14?

> @Pyrogenez Hello. Once the hat is attached to the raspberry pi powering pumps and raspberry pi is happening through hats J14 (labelled as power). You need to provide there 5V only. No need a USB C powering the raspberry pi at all. I hope this helps. Let me know if you have another question.

ipatiris, 2025-05-15:
@Pyrogenez Hello. I am using the HUBS238 as suggested to this project. I have a raspberry pi power supply powering the HUBS238 and from there to the hat. Setup looks like:
[image]

Pyrogenez, 2025-05-15:
Oh wow, I'm an idiot lol I forgot about the husb!!  I'm just getting over a week+ with cocid/strep throat so I'm going to blame my Idiocracy on the sickness LOL 

Thank you!!! 

> @Pyrogenez Hello. I am using the HUBS238 as suggested to this project. I have a raspberry pi power supply powering the HUBS238 and from there to the hat. Setup looks like:
> [image]

Pyrogenez, 2025-05-16:
@ipatiris - I notice you left the 1A jumper closed.  Are you noticing any issues?  I opened all amp jumpers so it sends 3A but I am still experiencing issues with low power.  The RPi red light turns off with any action.  Maybe there is a short that I've yet to find. 

> [@Pyrogenez](https://github.com/Pyrogenez) Hello. I am using the HUBS238 as suggested to this project. I have a raspberry pi power supply powering the HUBS238 and from there to the hat. Setup looks like: [image]

imbahzor, 2025-08-01:
> [@ipatiris](https://github.com/ipatiris) - I notice you left the 1A jumper closed. Are you noticing any issues? I opened all amp jumpers so it sends 3A but I am still experiencing issues with low [...]

Did you ever solve this? Experiencing the same issue

Pyrogenez, 2025-10-08:
No, I never resolved this. I still have issues with power loss. 

> Did you ever solve this? Experiencing the same issue