Concept-Bytes/Tipsy #18: flow rate carbonated ingredients

Hemuc71, 2025-06-10:
I have a problem with the pumps: with carbonated ingredients (like tonic water and soda), the flow rate is very limited. I've already tried using slightly larger tubing. Is anyone else experiencing this issue, and does anyone possibly know what the solution could be? I’ve already ordered a pump with larger tubing (3x5 mm and 4x6 mm). Any other tips are very welcome!

By the way wonderfull project super :-) Thanks for all the work on this.

tetra76, 2025-06-23:
i´ve changed 4 Pumps to Membrane Pumps (https://amzn.eu/d/9eM1ODJ) for Carbonated Fluids (Coke, Soda, Prosecco and Red-Bull). The only negative is that you can't use the cleaning function because the flow direction can't be reversed. I flush the pumps in the specified direction after use.

tetra76, 2025-08-13:
> Hey @tetra76 ,
> did you end up changing the OZ-Coefficient since you then had two differend pump types with different flow rates ?
> Im currently thinking of putting 6x peristaltic pumps and 6x membrane pumps in my setup but im struggling with finding precise pumps with the correct flow rate. Also i think only 6v pumps work with the pcb. do you know if im right?

Hey Audiopicasso,I‘ve tried some settings, but with different types (soda, prosecco etc) you need to test the settings again. The carbon in soda breaks the flow more than in prosecco. I try actual different types of push methodes. (Closed bottles with a air pushing pump and overpressure push the soda with no loss of carbon)I use 6v Pumps and cut the power line from the pi-tipsy hat to gpio and powered only the pump switch side, so i can use also higher voltage like 7,5 or 9volt. But you need a second voltage for the raspi (i have a 3B)